Xinyuan-LilyGO/T-ZigBee #1: Thread border router

jonsmirl, 2022-04-27:
This hardware is capable of being a Thread border router. Are you working on the software for doing that?

liangyingy, 2022-04-28:
The original factory does not provide thread sdk. There is no plan to develop a thread for the time being.

http://wiki.telink-semi.cn/wiki/chip-series/TLSR825x-Series/

[image]

jonsmirl, 2022-04-28:
Their datasheet says they have it:
http://wiki.telink-semi.cn/doc/ds/PB_TLSR8258-E_Product%20Brief%20for%20Telink%20BLE%20IEEE802.15.4%20Multi-Standard%20Wireless%20SoC%20TLSR8258.pdf

6LoWPAN and Thread features
 Supports 6LowPAN, IPv6 and DHCPv6
 Supports UDP and DTLS
 Supports thread security and commission

So it should be coming

jonsmirl, 2022-04-28:
The Thread implementation may be inside the HomeKit support.

liangyingy, 2022-04-28:
The HomeKit sdk of tlsr8258 is based on ble.

jcomas, 2022-04-29:
Here http://wiki.telink-semi.cn/wiki/protocols/Zigbee/ you find Zigbee SDK. Then, you can use the modifications from https://github.com/pvvx/TlsrComSwireWriter to upload a new firmware in TLSR8258.

liangyingy, 2022-05-10:
> Can you tell me more about the included Zigbee firmware? Maybe, it is a similar to "tl_zigbee_sdk\apps\sampleGw" from Zigbee SDK. I'm interested in comunicate ESP via UART to TLSR8258 using HCI (zbhci.h), but I'd like TLSR8258 as Zigbee enddevice.

You can refer to [light_demo](https://github.com/Xinyuan-LilyGO/T-ZigBee/tree/main/examples/light_demo).

closed 2022-11-04 by liangyingy