Xinyuan-LilyGO/T-ZigBee #3: [REQUEST] Telink radio library  for zigpy + a Zigbee serial API proxy/bridge mode?

Hedda, 2022-05-03:
Any chance you would consider adding a Zigbee serial API proxy/bridge and making a Telink radio library (in Python) for zigpy?

https://github.com/zigpy/zigpy

https://github.com/zigpy/zigpy-cli

zigpy is an open-source Python library implementing Zigbee serial API hardware-independent abstraction for radio manufacturers and many popular projects depend on zigpy, including ZHA for Home Assistant, Zigbee plugin for Domoticz, and Zigbee plugin for Jeedom:

https://www.home-assistant.io/integrations/zha

https://www.domoticz.com/wiki/ZigbeeForDomoticz

https://doc.jeedom.com/en_US/plugins/automation%20protocol/zigbee/

So all and any Zigbee Coordinator adapters that have a radio library for zigpy can be used with ZHA, Domoticz, and Jeedom:

https://github.com/zigpy/zigpy-znp

https://github.com/zigpy/zigpy-deconz

https://github.com/zigpy/bellows

https://github.com/zigpy/zigpy-zigate

https://github.com/zigpy/zigpy-xbee

liangyingy, 2022-05-27:
Is there any relevant zigpy documentation? I did not find it

Hedda, 2022-05-30:
@liangyingy those zigpy libraries is a free and open-source projects developed by volunteers from the DIY home automation software communities in their spare time so unfortunately there is not a lot of documentation, instead if you (or Telink) are interested in taking on this project you would primarily have to look at the existing radio libraries like the [zigpy-znp](https://github.com/zigpy/zigpy-znp) radio library (alternatively [bellows](https://github.com/zigpy/bellows) or [zigpy-deconz](https://github.com/zigpy/zigpy-deconz/)) to see how the hardware API translation is done from proprietory API to the zigpy radio API which is meant to be unified for all zigpy radio APIs. Then of couse create and issue or discussion to post question on specific problems if get stuck. 

For reference you can see the somewhat similar discussion here about creating a new radio library for zigpy-> https://github.com/zigpy/zigpy/issues/394

Note! @puddly is working on updating the zigpy radio API making all radio libraries more unified, see -> https://github.com/zigpy/zigpy-cli/pull/2

Therefore if starting before those changes are merged it would currently probably be best to look at his development branches: 

https://github.com/puddly/zigpy-cli/tree/puddly/zigpy-radio-api

https://github.com/puddly/zigpy/tree/puddly/new-radio-settings-api

https://github.com/puddly/zigpy-znp/tree/puddly/new-radio-settings-api

https://github.com/puddly/bellows/tree/puddly/zigpy-radio-api

https://github.com/puddly/zigpy-deconz/tree/puddly/new-radio-api

Hedda, 2022-08-03:
FYI, puddly has now merged the new radio API so all all of zigpy radio libraries should now use the new unfied API.

Hedda, 2023-04-28:
Tip on another possible alternative ESP32 target hardware for that described scenario could be the **Espressif  ESP32-C6 SoC series** which add an integrated 802.15.4 radio that supports Zigbee 3.0 (and Thread). See related discussion here -> https://github.com/zigpy/zigpy/discussions/783

PS: I believe that those with Espressif contacts can get such DevKit boards for evaluation upon request(?) -> [sales@espressif.com](mailto:sales@espressif.com)

lbuque, 2023-04-28:
@Hedda You can check this discussion #22, I made a light device using esp32-c6 and successfully added it to home assistant.