Xinyuan-LilyGO/T-ZigBee #4: ESPHome or tasmota supported?

robgardien, 2022-05-12:
Is it possible to install ESPHome or tasmota on this device? Ik woud like to connect a 3.3 volt light and use this in Home Assistant

pawilli, 2022-12-30:
> Is it possible to install ESPHome or tasmota on this device? Ik woud like to connect a 3.3 volt light and use this in Home Assistant

> 

Yes it is possible to install [ESPHOME](https://esphome.io/index.html) on the T-zigbee ESP32C3.  Use this code to set up a simple light example

```
esphome:
  name: esp32c3

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

#Enable logging
logger:

#Enable Home Assistant API
api:
  encryption:
    key: "IT WILL GENERATE A KEY FOR YOU"

ota:
  password: "IT WILL GENERATE AN OTA PASSWORD FOR YOU"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

#Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32C3 Fallback Hotspot"
    password: "IT WILL GENERATE A PASSWORD FOR YOU"

captive_portal:

#Simple light using code from ESPHOME https://esphome.io/components/light/status_led.html
light:
  - platform: status_led
    name: "TZB LED"
    pin: GPIO3
```

This light can be controlled from HA and also from Alexa if you have the Alexa integration set up.

16:54:20][D][light:035]: 'TZB LED' Setting:
[16:54:20][D][light:046]:   State: ON
[16:54:20][D][status_led:051]: 'TZB LED': Setting state ON
[16:54:22][D][light:035]: 'TZB LED' Setting:
[16:54:22][D][light:046]:   State: OFF
[16:54:22][D][status_led:051]: 'TZB LED': Setting state OFF
[16:54:28][D][light:035]: 'TZB LED' Setting:
[16:54:28][D][light:046]:   State: ON
[16:54:28][D][status_led:051]: 'TZB LED': Setting state ON
[16:54:40][D][esp32.preferences:113]: Saving 1 preferences to flash...
[16:54:40][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[16:54:46][D][light:035]: 'TZB LED' Setting:
[16:54:46][D][light:046]:   State: OFF
[16:54:46][D][status_led:051]: 'TZB LED': Setting state OFF

TiddlyWiddly, 2022-12-31:
What about the Zigbee piece of it?

teras, 2024-01-05:
I am curious, is there any more info on this? I have the same question. I have successfully installed esphome web server, but I am curious if I can use the zigbee part of it  (which is actually the main reason why someone would buy something like this)