Xinyuan-LilyGO/T-ZigBee #5: Better flashing instructions

yetanothercarbot, 2022-05-17:
I received my T-Zigbee boards today. I soldered headers onto one and connected it via USB with the dip switch set to the C3; it failed to enumerate (in fact, it knocked my USB headphones off the bus!).

IO9 isn't broken out so I'm unsure how to place the ESP32-C3 into download mode so that I can flash new firmware. 

I have other questions too, but closing #2 should solve those.

liangyingy, 2022-05-17:
need to use u2t:

[image]

yetanothercarbot, 2022-05-17:
I see. That's unfortunate; the price of shipping is nearly the price of the U2T itself haha. Thanks for the quick reply.

camnop, 2022-06-04:
Is there any way to flash using a FT232RL USB TO TTL? If yes, how to connect the wires?

liangyingy, 2022-06-06:
@camnop 

You can use [USBCOMFlashTx.html](https://github.com/pvvx/TlsrComSwireWriter/blob/master/USBCOMFlashTx.html) in https://github.com/pvvx/TlsrComSwireWriter to flash the TLSR8258 firmware.

**But FTDI USB-COM chips don't work!**

You can use t-u2t or ch340.

Below the connection diagram：

[image]

camnop, 2022-06-06:
What about PL2303HX using 3v?

[image]

[image]

liangyingy, 2022-06-07:
@camnop This PL2303HX has no RTS.

liangyingy, 2022-06-08:
[Burning Guide](https://zbhci.readthedocs.io/en/latest/user-guide/burning.html)
[烧录指南](https://zbhci.readthedocs.io/zh_CN/latest/user-guide/burning.html)

camnop, 2022-06-11:
Thanks! I got the board recognized using this Downloader Board:
[image]

I flash it with [sampleGW_8258_20220302.bin](https://github.com/Xinyuan-LilyGO/T-ZigBee/blob/main/firmware/sampleGW_8258_20220302.bin)

following your burning guide.

closed 2022-08-05 by liangyingy