Xinyuan-LilyGO/T-ZigBee #6: Zigbee2MQTT - Home Assistant - What to flash?

camnop, 2022-06-11:
How to flash Zigbee2MQTT?  I didnt get what i have to do to come in this kind of configuration:

https://zbhci.readthedocs.io/en/latest/user-guide/zigbee2mqtt.html

What I have to flash? 

My goal is to use zigbee2mqtt in Home Assistant

liangyingy, 2022-06-11:
Burn this program to esp32-c3: https://github.com/Xinyuan-LilyGO/T-ZigBee/tree/main/examples/zigbee2mqtt

camnop, 2022-06-13:
How to flash it? I tried to flash using platformio in windows but a received error.. 

I will try to attach the evidence here  later

camnop, 2022-06-18:
Hi @liangyingy, need I press some button when I'm flashing it?

liangyingy, 2022-06-20:
Please refer to the programming guide https://zbhci.readthedocs.io/en/latest/user-guide/burning.html

camnop, 2022-06-20:
@liangyingy what about using ch340? Need I press some button?

swkim01, 2022-06-24:
Hello, I made a mistake to erase flash before burning zigbee2mqtt firmware. Thus, the board failed to mount littleFS.
> ./components/esp_littlefs/src/littlefs/lfs.c:1071:error:` Corrupted dir pair at {0x0, 0x1}
> E (193) esp_littlefs: mount failed,  (-84)
> E (193) esp_littlefs: Failed to initialize LittleFS

I guess that the file system for web configuration were erased. So, I'm curious how to make the FS and its contents.

liangyingy, 2022-06-24:
> Hello, I made a mistake to erase flash before burning zigbee2mqtt firmware. Thus, the board failed to mount littleFS.
> 
> > ./components/esp_littlefs/src/littlefs/lfs.c:1071:error:` Corrupted dir pair at {0x0, 0x1}
> > E (193) esp_littlefs: mount failed,  (-84)
> > E (193) esp_littlefs: Failed to initialize LittleFS
> 
> I guess that the file system for web configuration was erased. So, I'm curious how to make the FS and its contents.

You need to use platformio to do the following:

[image]

We will be compatible with arduino ide in the future. Because there is currently no plug-in suitable for arduino ide to create and upload the file system

swkim01, 2022-06-24:
> You need to use platformio to do the following:
>
> [image]
>
> We will be compatible with arduino ide in the future. Because there is currently no plug-in suitable for arduino ide to create and upload [...]

Now I can upload the file system via arduino ide! I used the Esp32 sketch data upload menu on arduino ide by installing [arduino-esp32fs-plugin](https://github.com/lorol/arduino-esp32fs-plugin). Anyway, I appreciate your effort. Thank you.

ejri, 2022-08-06:
Hey,

@camnop, did you end up uploading the example onto the esp32?

@liangyingy 
When I flash it, it seems to disconnect.  

Setup:
one button connected to GPIO04 and GND. 
MQTT broker + port updated to my broker 
wifi and password updated to my network

[image]

swkim01, 2022-08-11:
@ejri I used recent released one of [arduino-esp32fs-plugin](https://github.com/lorol/arduino-esp32fs-plugin/releases) and selected LittleFS format on the dialog window, not SPIFFS.
No problem with 2.0.2 or 2.0.3 versions of esp32 arduino core.

pcopa, 2022-09-25:
@liangyingy : I am confused. In your text you say "don't use Arduino", but in your examples you DO show "Arduino".
I only use PlatformIO, but don't get joy. 
All Builds and Uploads have no errors. See As mentioned in #13 I get:

` Executing task in folder zigbee2mqtt: C:\Users\xxxxx.platformio\penv\Scripts\platformio.exe device monitor

--- Terminal on COM7 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32c3-api1-20210207
Build:Feb 7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
./components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
E (211) esp_littlefs: mount failed, (-84)
E (211) esp_littlefs: Failed to initialize LittleFS
[ 210][E][LittleFS.cpp:94] begin(): Mounting LittleFS failed! Error: -1
An Error has occurred while mounting LittleFS
`

ejri, 2022-10-04:
@camnop yeah I think so too. I tried flashing and a few different combinations on 3 boards. 

The probability of 1 broken board may be low.
The probability of 2 broken boards is even lower.
3 broken boards probably means there's something wrong with the board design or the firmware. 

I think I'll wait for future iterations of the board. It would be amazing the board eventually works given the unique integration between esp32 and zigbee.

pawilli, 2022-11-27:
> @ejri @pcopa @swkim01 I recorded what I did... I start to think that we don't have any example project that really works...
> 
> #16 https://youtu.be/OmtmtGZHcNg

I can confirmed that both of the following examples work with some slight modifications.

1. **Light demo - simple light example** to set up a simple light that you can use to control the onboard led but more importantly renders the board as a cheap but efficient ZigBee router to extend your ZigBee network
-> Make sure you flash the TLSR8258 with the correct firmware - [hciDevice_ZR_8258_SDK_eb644f1_MD5_3e7c395bacca28475952808a79ff2ec1.bin](https://github.com/Xinyuan-LilyGO/T-ZigBee/blob/main/firmware/hciDevice_ZR_8258_SDK_eb644f1_MD5_3e7c395bacca28475952808a79ff2ec1.bin)
-> Read this readme file in its entity to save some time [https://github.com/Xinyuan-LilyGO/T-ZigBee/blob/main/examples/light_demo/README.md].  I worked on this example for hours before realizing I need a different firmware - see above.
 

4. Zigbee2MQTT example and paired a Xiaomi RTCGQ11LM that is then integrated into a test set up for home assistant and can accurately detect motion and readout the lux value - see image.  I have also paired the simple light above and it is controllable from HA. 

 I also paired a Sonoff plug but it is not reflected in the mqtt as it may not have the definitions in the device.c and device.h files.

-> you might need to hardcode your credentials into the ...\data\db.json especially if your mqtt server needs authentication
-> use the latest git versions from @lbuque as it contains an updated ...\zigbee2mqtt\app_mqtt.cpp that allows authentication.
-> After you install the new versions run the following from the terminal in platformio after you clean and build the new platformio.ini  to re-flash littlefs
```
pio run -e zigbee2mqtt -t upload
pio run -e zigbee2mqtt -t uploadfs

```
[image]

@lbuque please let us know how to add new devices or submit them for inclusion

Thanks
Des

lbuque, 2022-12-05:
@pawilli Documentation for adding new devices is still in preparation and is currently busy with other projects.

Tircson, 2024-03-20:
Hello,
after flashing the Firmeware for the TLSR8258 do I need to reset the dip back to ESP32-Mode? After following the instructions for 
 the zigbee2mqtt i get the following error

```
Serial port COM3
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-C3: Serial data stream stopped: Possible serial noise or corruption.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 2
```

And if i do it with the dip switches in the ESP32-Mode i get the 
```
./components/esp_littlefs/src/littlefs/lfs.c:1071:error:` Corrupted dir pair at {0x0, 0x1}
E (193) esp_littlefs: mount failed, (-84)
E (193) esp_littlefs: Failed to initialize LittleFS

```
error