Xinyuan-LilyGO/T-ZigBee #7: Light demo better instructions

Math1010, 2022-07-02:
Hello,
Is it possible to have more explanation about the light demo application which button press to go in pairing mode ? 
Have you an example of the external converter file used in zigbee2MQTT ? 

Thanks you !

liangyingy, 2022-07-07:
added: https://github.com/Xinyuan-LilyGO/T-ZigBee/tree/main/examples/light_demo

swkim01, 2022-07-14:
In case of using wired coordinator such as sonoff zigbee 3.0 usb dongle plus, the following switch converter would work as to [this procedure](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#instructions).

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LILYGO.Light'],
    model: 'LILYGO.Light',
    vendor: 'LilyGO',
    description: 'light_demo',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [e.switch()],
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```
screenshot of z2m device list on HA.
[image]
screenshot of z2m dashboard on HA.
[image]

Math1010, 2022-07-21:
Thanks 
I have a CC2652 usb stick on a raspberry pi 4 with home assistant and Zigbee2MQTT and everything is working.

closed 2022-07-21 by Math1010

ejri, 2022-08-06:
@liangyingy  @Math1010 would you please provide better instructions?

What are the steps followed? just upload light_demo.ino?
How does the board enter pairing mode? 

Long press doesn't seem to work. The board doesn't get recognized by Zigbee2MQTT on HA.
Single press returns: "Not joined the zigbee network"