Xinyuan-LilyGO/T-ZigBee #11: New sensor demo application with dht sensor

swkim01, 2022-08-01:
Hello,

I added new sensor demo application in my repository. You can test it with a dht11/22 sensor. Enjoy!
[https://github.com/swkim01/T-ZigBee/tree/main/examples/sensor_demo](https://github.com/swkim01/T-ZigBee/tree/main/examples/sensor_demo)

liangyingy, 2022-08-01:
great! Welcome to submit PR!

liangyingy, 2022-08-01:
#12

closed 2022-08-01 by liangyingy

pawilli, 2023-01-03:
> Hello,
> 
> I added new sensor demo application in my repository. You can test it with a dht11/22 sensor. Enjoy! https://github.com/swkim01/T-ZigBee/tree/main/examples/sensor_demo

Hi @swkim01.  

Thank you for working on the sensory demo.  I have a DS18B20 and can get it to work using on this board using ESPhome but the esphome implementation is unstable and I would prefer to pick up the temperature through your zigbee-based program.  

How can I change the DHT in the example you provided to a DS18B20?

Thanks