Xinyuan-LilyGO/T-ZigBee #13: TLSR8258 not responding after upgrade (Red Led PD4)

SangaranSiva, 2022-08-08:
Hello,

I try to use the sensor_demo example to use my devices to connect to Zigbee2MQTT.
I upgrade the firmware of TLSR8258 following the instructions (DIP switch and U2T) on sensor_demo, I upload the sensor_demo code and there was no error when flashing but the red led (PD4) is not working anymore.

I don't know if it's normal or not.
Is there something I have to do ?

[image]

ejri, 2022-08-09:
Had the same issue. No red light after TLSR8258 firmware update. 

Though the sensor_demo doesn't seem to work for me...

swkim01, 2022-08-09:
@SangaranSiva @ejri How did you build and upload the firmware from the sensor_demo.ino?
Use arduino ide in order to build and upload the firmware with esp32-c3 settings as [this link](https://zbhci.readthedocs.io/en/latest/user-guide/burning.html).

SangaranSiva, 2022-08-09:
@swkim01 I upgraded the firmware of TLSR8258 to [hciDevice_ZR_8258_SDK_eb644f1_MD5_3e7c395bacca28475952808a79ff2ec1.bin](https://github.com/Xinyuan-LilyGO/T-ZigBee/blob/main/firmware/hciDevice_ZR_8258_SDK_eb644f1_MD5_3e7c395bacca28475952808a79ff2ec1.bin) using USBCOMFlashTx.html found in the same [link](https://zbhci.readthedocs.io/en/latest/user-guide/burning.html).

Ok I will try, thanks.

ejri, 2022-08-09:
Yeah, followed the same link you posted. 

factory_test.ino works as expected, http request over wifi. Though for anything related to zigbee, it doesn't seem to work. 

zigbee2mqtt returns a LittleFS error, saying that the board is a esp32-c3 chip, despite choosing the esp32-c3 board.
https://github.com/Xinyuan-LilyGO/T-ZigBee/issues/6

light_demo.ino doesn't get recognized by sonoff 3.0 usb 
https://github.com/Xinyuan-LilyGO/T-ZigBee/issues/7

liangyingy, 2022-08-10:
@ejri @SangaranSiva

Please follow this document strictly to burn new firmware for tlsr8258.

Make sure that the Red Led PD4 is working properly, and then perform other operations.

ejri, 2022-08-10:
to clarify, I followed the document very closely. 

This is a screenshot that the firmware file has been successfully uploaded, as in the document. 

[image]

liangyingy, 2022-08-10:
This flasher has no feedback, it is only responsible for sending, no matter whether the chip is receiving normally or not.

liangyingy, 2022-08-10:
Before burning tlsr8258, you need to burn [factory_test](https://github.com/Xinyuan-LilyGO/T-ZigBee/tree/main/examples/factory_test).

factory_test is used to power on the tlsr8258

SangaranSiva, 2022-08-11:
I also try as [mention](https://github.com/pvvx/TlsrComSwireWriter) here using python, but the led is still off. 
```
python3 TLSR825xComFlasher.py -p /dev/cu.usbmodem539C0127771  -t 70 wf 0 hciDevice_ZR_8258_SDK_eb644f1_MD5_3e7c395bacca28475952808a79ff2ec1.bin
=======================================================
TLSR825x Flasher version 00.00.04
-------------------------------------------------------
Open /dev/cu.usbmodem539C0127771, 921600 baud...
Reset module (RTS low)...
Activate (70 ms)...
UART-SWS 92160 baud. SW-CLK ~24.9 MHz(?)
Inputfile: hciDevice_ZR_8258_SDK_eb644f1_MD5_3e7c395bacca28475952808a79ff2ec1.bin
Write Flash data 0x00000000 to 0x00032f84...
-------------------------------------------------------
Worked Time: 81.594 sec
Done!```

liangyingy, 2022-08-15:
> I also try as mention [here](https://github.com/pvvx/TlsrComSwireWriter) using python, but the led is still off. It looks like the tlsr8258 is bricked.
> 
> ```
> python3 TLSR825xComFlasher.py -p /dev/cu.usbmodem539C0127771  -t 70 wf 0 hciDevice_ZR_8258_SDK_eb644f1_MD5_3e7c395bacca28475952808a79ff2ec1.bin
> =======================================================
> TLSR825x Flasher version 00.00.04
> -------------------------------------------------------
> Open /dev/cu.usbmodem539C0127771, 921600 baud...
> Reset module (RTS low)...
> Activate (70 ms)...
> UART-SWS 92160 baud. SW-CLK ~24.9 MHz(?)
> Inputfile: hciDevice_ZR_8258_SDK_eb644f1_MD5_3e7c395bacca28475952808a79ff2ec1.bin
> Write Flash data 0x00000000 to 0x00032f84...
> -------------------------------------------------------
> Worked Time: 81.594 sec
> Done!
> ```

Do you have a win computer? You can try to use a win computer to burn the firmware!

At present, we have not received feedback from other customers that tlsr8258 cannot be burned!

Hope to get your help! Find out if it is a product defect! ! !

SangaranSiva, 2022-08-18:
Hi, I try using a win computer, but the led is still off.
```
python3 TLSR825xComFlasher.py -p COM18  -t 70 wf 0 hciDevice_ZR_8258_SDK_eb644f1_MD5_3e7c395bacca28475952808a79ff2ec1.bin
=======================================================
TLSR825x Flasher version 00.00.04
-------------------------------------------------------
Open COM18, 921600 baud...
Reset module (RTS low)...
Activate (70 ms)...
UART-SWS 92160 baud. SW-CLK ~23.5 MHz(?)
Inputfile: hciDevice_ZR_8258_SDK_eb644f1_MD5_3e7c395bacca28475952808a79ff2ec1.bin
Write Flash data 0x00000000 to 0x00032f84...
-------------------------------------------------------
Worked Time: 85.042 sec
Done!
```

liangyingy, 2022-08-18:
You use a multimeter to measure the two places marked in Chinese in the picture below.

Is the voltage 3.3V

[image]

SangaranSiva, 2022-08-18:
Yes it's 3.3V.

liangyingy, 2022-08-18:
Is the status of the dip switch the same as the picture below?

[image]

[image]

SangaranSiva, 2022-08-18:
Yes it's the same.

liangyingy, 2022-08-19:
Please contact our customer service for compensation!

ejri, 2022-08-19:
It’s the same for me for one of the boards. 

I have 3 boards in total. I’ll try it on all the other two and let you know.

ejri, 2022-08-22:
@liangyingy I have tried the three boards and they have the same issue. I will contact customer service and share with them the video of uploading the firmware and testing it, exactly as mentioned in https://zbhci.readthedocs.io/en/latest/user-guide/burning.html

liangyingy, 2022-08-22:
@ejri I don't quite believe that all three boards will have the same problem.

ejri, 2022-08-22:
Sent the video to customer service, please check it. (it's a youtube video showing the step by step, following the tutorial).

pcopa, 2022-09-24:
I'm having the same issue.  Tried to flash several images but they all result in "Failed to initialize LittleFS".
This is what I get:

` Executing task in folder zigbee2mqtt: C:\Users\schel\.platformio\penv\Scripts\platformio.exe device monitor 

--- Terminal on COM7 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
./components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
E (211) esp_littlefs: mount failed,  (-84)
E (211) esp_littlefs: Failed to initialize LittleFS
[   210][E][LittleFS.cpp:94] begin(): Mounting LittleFS failed! Error: -1
An Error has occurred while mounting LittleFS
`

liangyingy, 2022-09-26:
@pcopa 
Please execute the following command in the terminal of platformio:
```
pio run -e ttgo-t-oi-plus -t upload
pio run -e ttgo-t-oi-plus -t uploadfs
```

shojkeee, 2022-11-12:
The LED does not work because the instructions for the sensor_demo and light_demo firmware present the same firmware for TLSR8258. I think that's the problem. Please people who are running this git, edit the instructions. It seems to me that the wrong firmware has been added for sensor_demo

shojkeee, 2022-11-12:
but no. as it turned out, this is not the problem. liligo.light is working. but sensor_demo is not. As soon as sensor_demo.ino is poured into the esp32-c3, the diodes stop burning from the TLSR microcontroller. The problem is clearly sensor_demo.ino

shojkeee, 2022-11-12:
I found what the problem is. The sensor_demo program is not going correctly because of the platform.ini file. Change platform.ini to the values that I attach below in the sensor_demo folder

[platformio]
src_dir = .
lib_dir = ../..

[env:ttgo-t-oi-plus]
platform = espressif32
board = ttgo-t-oi-plus
framework = arduino
board_build.mcu = esp32c3
board_build.f_cpu = 160000000L
upload_speed = 921600
monitor_speed = 115200
lib_deps = 
	mathertel/OneButton@^2.0.3
	adafruit/DHT sensor library@^1.4.4
	adafruit/Adafruit Unified Sensor@^1.1.5
build_flags = -D CORE_DEBUG_LEVEL=3

lbuque, 2022-11-14:
@shojkeee The following code snippet has enabled RT9080:
https://github.com/Xinyuan-LilyGO/T-ZigBee/blob/42fd137ad32bba083e7f0f41dcc42219afecbdde/examples/sensor_demo/sensor_demo.ino#L46-L47

milekium, 2023-06-26:
Before flashing the firmware with 'TLSR825xComFlasher  wf  '  you need to delete all regions of flash with 'TLSR825xComFlasher ea '.

purha, 2023-08-31:
I have the same issue, after flashing the TRLS chip, the red light doesnt turn on, none of the TLSR825xComFlasher.py commands work.
Before using the web ui to flash I tried python version which gave the error:

```
Reset module (RTS low)...
Activate (6000 ms)...
UART-SWS 92160 baud. SW-CLK ~43.3 MHz(?)
Inputfile: hciDevice_ZE_8258_SDK_eb644f1_MD5_a27ce68a94bc19284c5c87a4e0aa416b.bin
Write Flash data 0x00000000 to 0x00027404...
Error Read Flash Status! (0)
Error write sector at 0x000100!
```

and after flashing from the ui the device seems to be bricked.

```
❯ python3 TLSR825xComFlasher.py -p /dev/cu.usbserial-54350178781 -t 3000 ea
=======================================================
TLSR825x Flasher version 00.00.04
-------------------------------------------------------
Open /dev/cu.usbserial-54350178781, 921600 baud...
Reset module (RTS low)...
Activate (3000 ms)...
Chip sleep? -> Use reset chip (RTS-RST): see option --tact
```

any ideas?

asat232, 2024-09-22:
<b>DO NOT flash TRLS chip on MacOS.</b>
I had the same issue after flashing TRLS chip on MacOS (Intel), but i was lucky to make it works again after flashing on Windows.

lbuque, 2024-09-22:
这是来自QQ邮箱的自动回复邮件。
 
你好，我已经收到您的邮件。我会尽快查看并给你回复。