Xinyuan-LilyGO/T-ZigBee #14: Incorrect information for ZBHCI_CMD_DATA_CONFIRM

wes58, 2022-09-09:
When I receive ZBHCI_CMD_DATA_CONFIRM message I get the following:
hci_uart: 0x3fc97bb0   55 82 00 00 09 2b 02 36  b5 01 0b 00 19 00 32 aa  
zbhci: Type: 0x8200
zbhci:   (Data Confirm)
zbhci:   Status:  0x36
zbhci:   Src Ep:  0x2
zbhci:   Aps Cnt: 0xb5

This is incorrect.
Telink ZGC TOOL shows the following:
22-09-09 08:22:21.544841 recv<--:55 82 00 00 09 c7 02 36 b5 01 0b 00 19 00 a4 aa 
hci commandid: 0x8200 (ZBHCI_CMD_DATA_CONFIRM)
payload len: 9
	dstAddr_mode:0x02(APS_SHORT_DSTADDR_WITHEP)
	dst_addr: 0x36b5
	src_endpoint: 0x01
	dst_endpoint: 0x0b
	cluster_id: 0x0019(OTA)
	status:0x0(SUCCESS)
	aps_cnt:0xa4	

This makes more sense. Can you please have a look at this?

liangyingy, 2022-09-09:
But telink sdk only sends `Status`, `Src Ep` and `Aps Cnt`.

[image]

wes58, 2022-09-09:
> From "liangyingy" ***@***.***>
> To "Xinyuan-LilyGO/T-ZigBee" ***@***.***>
> Cc "wes58" ***@***.***>; "Author" ***@***.***>
> Date 9/09/2022 11:06:55
> Subject Re: [Xinyuan-LilyGO/T-ZigBee] Incorrect information for
> ZBHCI_CMD_DATA_CONFIRM (Issue #14)

Hi,
They must have fixed it in the latest SDK v3.6.8.
       if(pApsDataCnf->dstAddrMode == APS_DSTADDR_EP_NOTPRESETNT){
             *pBuf++ = pApsDataCnf->srcEndpoint;
         }else if(pApsDataCnf->dstAddrMode == APS_SHORT_GROUPADDR_NOEP){
             COPY_U16TOBUFFER_BE(pBuf, pApsDataCnf->dstAddr.addr_short);
             pBuf += 2;
             *pBuf++ = pApsDataCnf->srcEndpoint;
         }else if(pApsDataCnf->dstAddrMode == APS_SHORT_DSTADDR_WITHEP){
             COPY_U16TOBUFFER_BE(pBuf, pApsDataCnf->dstAddr.addr_short);
             pBuf += 2;
             *pBuf++ = pApsDataCnf->srcEndpoint;
             *pBuf++ = pApsDataCnf->dstEndpoint;
         }else if(pApsDataCnf->dstAddrMode == APS_LONG_DSTADDR_WITHEP){
             ZB_IEEE_ADDR_REVERT(pBuf, pApsDataCnf->dstAddr.addr_long);
             pBuf += 8;
             *pBuf++ = pApsDataCnf->srcEndpoint;
             *pBuf++ = pApsDataCnf->dstEndpoint;
         }
         COPY_U16TOBUFFER_BE(pBuf, pApsDataCnf->clusterId);
         pBuf += 2;
         *pBuf++ = pApsDataCnf->status;
         *pBuf++ = pApsDataCnf->apsCnt;

         zbhciTx(ZBHCI_CMD_DATA_CONFIRM, (u8)(pBuf - conf), conf);

lbuque, 2022-11-14:
Adapting SDK v3.6.8.2

https://github.com/Xinyuan-LilyGO/T-ZigBee/tree/adapt_3.6.8.2
https://github.com/Xinyuan-LilyGO/telink-zigbee-sdk/tree/3.6.8.2

This is the compiled tlsr8258 firmware:

[bin.zip](https://github.com/Xinyuan-LilyGO/T-ZigBee/files/10000089/bin.zip)

Any feedback is welcome!