Xinyuan-LilyGO/T-ZigBee #15: Issue with ZBHCI_CMD_NODES_JOINED_GET_RSP

wes58, 2022-09-16:
Here is the response using Telink ZGC Tool
recv<-- : 55 80 40 00 1a 6e 00 02 00 00 02 00 a4 c1 38 04 50 19 49 e5 c7 19 a4 c1 38 18 45 4e a8 5e b7 d9 aa 
	total_cnt:0x2
	start_idx:0x0
	list_count:0x2
	status:0x0(ZBHCI_MSG_STATUS_SUCCESS)
	ieee_addr:0xa4c13804501949e5 short_addr:0xc719	
	ieee_addr:0xa4c13818454ea85e short_addr:0xb7d9

And here is in the Gateway:
hci_uart: 55 80 40 00 1a 6e 00 02 00 00 02 00 a4 c1 38 04 50 19 49 e5 c7 19 a4 c1 38 18 45 4e a8 5e b7 d9 aa 
Type: 0x8040
  (Nodes Joined Get Rsp)
  Status:     00
  Total:      0x2
  Start Idx:  00
  List Count: 00

The messages are exactly the same, but as you can see, List Count is different and mac addresses are not displayed.
When I tried to replace List Count (u8ListCount) with Total (u8TotalCnt) I get mac addresses but they are not correct - see below:
   Mac Addr List:
     Node 0: 0x200a4c138045019
     Node 1: 0x49e5c719a4c13818
For start_idx they are using - u16 startIdx = (p[0] << 8) + p[1]; , not u8?

I think getting mac addresses and short addresses is pretty important.

liangyingy, 2022-09-22:
You are right, we will schedule a time to fix it.

wes58, 2022-09-22:
I am trying to use some other commands and noticed - so far - that the following are not correct (did Telink changed them?)
1. ZBHCI_CMD_ZCL_ATTR_READ_RSP
2. ZBHCI_CMD_DISCOVERY_NWK_ADDR_RSP and ZBHCI_CMD_DISCOVERY_IEEE_ADDR_RSP

And probably others that I haven't tried yet

commit d4e5015c (2022-10-06): fix/zbhci: Fix wrong information of ZBHCI_CMD_NODES_JOINED_GET_RSP

closed #15