Xinyuan-LilyGO/T-ZigBee #16: ERROR - Disconnected from WiFi access point [WiFi-event] event: 5  - Credentials to use in MQTT broker

camnop, 2022-10-01:
Hi, I finally bought T-U2T and I flashed succesfuly the T-Zigbee with zigbee2mqtt project using platformIO.

First I burned zigbee2mqtt in ESP32-C3, after I burned TLSR8258 [sampleGW_8258_20220302.bin](https://github.com/Xinyuan-LilyGO/T-ZigBee/blob/main/firmware/sampleGW_8258_20220302.bin)

I changed in zigbee2mqtt project files, my SSID, and password from my WIFI  and my MQTT broker (home assistant with mosquitto)

My device has the red light bright which means ZigBee is running. Also, the green light is bright, which allows zigbee sub-devices to join the gateway. It also has the blue light slow flash which means it has connected to WiFi (but I can't see it in my router), but it is not connected to mqtt server.

I need to use credentials (username and password) to connect in my MQTT server, how can I do it?

Also, I`m receiving this error below. Do you know what could be? 
 
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
entry 0x403ce000
WiFi: Set mode to WIFI_AP_STA
Waiting for WiFi
[WiFi-event] event: 11
WiFi access point  stopped
[WiFi-event] event: 10
WiFi access point started
WiFi: softAP has been established
WiFi: please connect to the LilyGo-EECC
HTTP server started
Nothing saved yet!
[WiFi-event] event: 5
Disconnected from WiFi access point
[WiFi-event] event: 5
Disconnected from WiFi access point
[WiFi-event] event: 5
Disconnected from WiFi access point
E (11310) esp-tls: [sock=49] select() timeout
E (11310) TRANSPORT_BASE: Failed to open a new connection: 32774
E (11310) MQTT_CLIENT: Error transport connect
E (11313) APP_MQTT: Last error reported from esp-tls: 0x8006
[WiFi-event] event: 5
Disconnected from WiFi access point
```

camnop, 2022-10-02:
I also follow this burning guide:  https://zbhci.readthedocs.io/en/latest/user-guide/burning.html

First, I installed [examples/factory_test](https://github.com/Xinyuan-LilyGO/T-ZigBee/tree/main/examples/factory_test) in ESP32-C3 using using T-U2T in platformIO and after that, I installed [sampleGW_8258_20220302.bin](https://github.com/Xinyuan-LilyGO/T-ZigBee/blob/main/firmware/sampleGW_8258_20220302.bin) in TLSR8258 using using T-U2T in Google Chrome + [TlsrComSwireWriter](https://github.com/pvvx/TlsrComSwireWriter). Both with success and I saw the red light fixed.

After that, I returned the DIP switch to the ESP32-C3 
[image]

I connected again in my computer, using T-U2T for the serial monitor. 

Looks like it is not find any SSID / WIFI.. what is going on? Is my board not working well? 
[image]

WIFI Scan Test
Results: 

PS: My board has two antennas connected and it is really close from the wifi2.4G

camnop, 2022-10-04:
[]([image]
)

zigbee2mqtt project doesn't work 
https://youtu.be/OmtmtGZHcNg

liangyingy, 2022-10-06:
The save button is used to save the current configuration information.
The update button is used to refresh the current configuration information and verify that the save button is correct to save the configuration information.

They seem to give you ambiguity.

[image]

liangyingy, 2022-10-06:
[image]

You can also modify the relevant configuration items at the 1 mark, and then click the button at the 3 mark to upload, and the configuration will take effect.

liangyingy, 2022-10-06:
There is a bug in setting mqtt using the web page, which has been fixed at present. see eff375bc96dd4ebabf9cbb29e7898dd4190ab44c.

closed 2022-10-11 by liangyingy

camnop, 2022-10-19:
I will try it tonight, thanks.

Btw, how can I connect in a mqtt server using credentials? I don't find any password and username fields.

Thanks

liangyingy, 2022-10-20:
No credentials required

swkim01, 2022-10-20:
> I will try it tonight, thanks.
> 
> Btw, how can I connect in a mqtt server using credentials? I don't find any password and username fields.
> 
> Thanks

If required, you can add those fields at [app_mqtt.cpp](https://github.com/Xinyuan-LilyGO/T-ZigBee/blob/main/examples/zigbee2mqtt/app_mqtt.cpp) as follows.

```
…
extern String mqtt_server;
extern uint32_t mqtt_port;
char username[16] = "mqtt";
char password[16] = "mqtt";
…
void mqtt_app_start(void)
{
    esp_mqtt_client_config_t mqtt_cfg = {
        // .uri = MQTT_BROKER_URL,
        .host = mqtt_server.c_str(),
        .port = mqtt_port,
        .username = username,
        .password = password
    };
…
```

camnop, 2022-10-20:
Now it is connected in MQTT and in my wifi connection. 

I'm trying to have AirQuality Zigbee in Home Assistant, MQTT integration there. 

I tried to pair the device and it looks in terminal that it was recognized but nothing happen to me in Home Assistant, need I to config something more?

`Reconnecting to COM3     Connected!
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
entry 0x403ce000
WiFi: Set mode to WIFI_AP_STA
Waiting for WiFi
[WiFi-event] event: 11
WiFi access point  stopped
[WiFi-event] event: 10
WiFi access point started
WiFi: softAP has been established
WiFi: please connect to the LilyGo-EECC
HTTP server started
Nothing saved yet!
[WiFi-event] event: 5
Disconnected from WiFi access point
[WiFi-event] event: 4
Connected to access point
[WiFi-event] event: 7
Obtained IP address: 192.168.0.142
E (11297) esp-tls: [sock=49] select() timeout
E (11297) TRANSPORT_BASE: Failed to open a new connection: 32774
E (11297) MQTT_CLIENT: Error transport connect
[ 11271][E][app_mqtt.cpp:203] log_error_if_nonzero(): [APP_MQTT] Last error reported from esp-tls: 0x8006
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x40058fb4  RA      : 0x42005490  SP      : 0x3fcb6400  GP      : 0x3fc8e400
TP      : 0x3fc94dcc  T0      : 0xa5a5a5a5  T1      : 0x3fc93c4c  T2      : 0xa5a5a5a5  
S0/FP   : 0x00000000  S1      : 0x3c0c1000  A0      : 0x0000000b  A1      : 0x3c0c143c
A2      : 0x00000015  A3      : 0x3c0c143c  A4      : 0x00000000  A5      : 0x0000000b  
A6      : 0x00000000  A7      : 0x00000000  S2      : 0x3c0c1000  S3      : 0x3c0c1000
S4      : 0x3c0c1000  S5      : 0x0000000b  S6      : 0x00000000  S7      : 0x00000003  
S8      : 0x00000005  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x3fc8e400  T4      : 0x00000000  T5      : 0x403899dc  T6      : 0x4200687e  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x0000000b
MHARTID : 0x00000000

Stack memory:
3fcb6400: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00098048 0xa5a5a5a5 0xfe6265df 0xcc86ecff
3fcb6420: 0x00000000 0x00000000 0x0000008e 0x00000000 0x00000044 0x00000000 0x00000000 0x00000000
3fcb6440: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6460: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6480: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb64a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x0020ffe2 0x00000001 0x0000001f 0x00000000
3fcb64c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb64e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6500: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6520: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x0020ffe4 0x00000001
3fcb6540: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6560: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6580: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb65a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb65c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb65e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6600: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6620: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6640: 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[... 31 lines cut ...]
[WiFi-event] event: 10
WiFi access point started
WiFi: softAP has been established
WiFi: please connect to the LilyGo-EECC
HTTP server started
[WiFi-event] event: 5
Disconnected from WiFi access point
[WiFi-event] event: 4
Connected to access point
[WiFi-event] event: 7
Obtained IP address: 192.168.0.142
E (11306) esp-tls: [sock=49] select() timeout
E (11306) TRANSPORT_BASE: Failed to open a new connection: 32774
E (11306) MQTT_CLIENT: Error transport connect
[ 11280][E][app_mqtt.cpp:203] log_error_if_nonzero(): [APP_MQTT] Last error reported from esp-tls: 0x8006`

Device:
[image]
[image]
[image]

Home Assistant:
[image]

[image]

liangyingy, 2022-10-21:
@camnop 

Currently only the following devices are supported:

[image]

camnop, 2022-10-21:
> @camnop
> 
> Currently only the following devices are supported:
> 
> [image]

Is there any guide or tip about how to support new devices?

Is possible to use the lilygo board like a remote adapter in zigbee2mqtt (Home Assistant)?  

https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_adapter.html#_2-b-configure-ser2net-4-0

```
Connect to a remote adapter
This how-to explains how to run Zigbee2MQTT with an adapter on a remote location. We will use ser2net for this which allows to connect to a serial port over TCP. In this way you can e.g. setup a Raspberry Pi Zero with the adapter connected while running Zigbee2MQTT on a different system. The instructions below have to be executed on the system where the adapter is connected to.
```

![image](https://user-images.githubusercontent.com/4583389/197093613-ef1e7ea8-c9df-4848-9746-274db82df048.png

liangyingy, 2022-10-21:
@camnop 
zigbee2mqtt.io supports ti and silicon coordinators, lilygo uses telink, their serial port protocols are different, and remote adapter cannot be used.

liangyingy, 2022-10-21:
@camnop 
Can you provide the sniffer file for the [AirQuality Zigbee](https://www.zigbee2mqtt.io/devices/TS0601_air_quality_sensor.html) device? So I can try to adapt it.

camnop, 2022-10-21:
Are you saying this code? 

If not, how can I have this sniffer file?

```
Core  0 register dump:
MEPC    : 0x40058fb4  RA      : 0x42005490  SP      : 0x3fcb6400  GP      : 0x3fc8e400
TP      : 0x3fc94dcc  T0      : 0xa5a5a5a5  T1      : 0x3fc93c4c  T2      : 0xa5a5a5a5  
S0/FP   : 0x00000000  S1      : 0x3c0c1000  A0      : 0x0000000b  A1      : 0x3c0c143c
A2      : 0x00000015  A3      : 0x3c0c143c  A4      : 0x00000000  A5      : 0x0000000b  
A6      : 0x00000000  A7      : 0x00000000  S2      : 0x3c0c1000  S3      : 0x3c0c1000
S4      : 0x3c0c1000  S5      : 0x0000000b  S6      : 0x00000000  S7      : 0x00000003  
S8      : 0x00000005  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x3fc8e400  T4      : 0x00000000  T5      : 0x403899dc  T6      : 0x4200687e  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x0000000b
MHARTID : 0x00000000

Stack memory:
3fcb6400: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00098048 0xa5a5a5a5 0xfe6265df 0xcc86ecff
3fcb6420: 0x00000000 0x00000000 0x0000008e 0x00000000 0x00000044 0x00000000 0x00000000 0x00000000
3fcb6440: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6460: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6480: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb64a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x0020ffe2 0x00000001 0x0000001f 0x00000000
3fcb64c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb64e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6500: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6520: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x0020ffe4 0x00000001
3fcb6540: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6560: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6580: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb65a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb65c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb65e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6600: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6620: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb6640: 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb6660: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb6680: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb66a0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb66c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb66e0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb6700: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb6720: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb6740: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb6760: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb6780: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb67a0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb67c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb67e0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: 0000000000000000
```

liangyingy, 2022-10-21:
@camnop The acquisition of sniffing files can be done through this tool.
https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

camnop, 2022-10-21:
Thanks, I will post it tomorrow

camnop, 2022-11-11:
HI @lbuque 

I sniffed zigbee traffic from TS0601 Air Quality Sensor - TuYa - 0xA41D - 0xcc86ecfffe6265df

[TS0601_air_quality_sensor_TuYa_0xA41D_0xcc86ecfffe6265df.zip](https://github.com/Xinyuan-LilyGO/T-ZigBee/files/9985815/TS0601_air_quality_sensor_TuYa_0xA41D_0xcc86ecfffe6265df.zip)

[image]

[image]

[image]

Do you think you can adapt it to work in T-Zigbee? May next time you cold give us a guide about how to adapt new devices.

lbuque, 2022-11-11:
This device uses the transparent transmission protocol. At present, I am not sure whether T-ZigBee can receive this transparent data (this transparent data may be discarded by the gateway and will not be sent to esp32-c3). I have a similar device on hand, I can try to adapt it.

If you want to know the specific definition of this transparent data, you can refer to the link: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

pawilli, 2022-11-17:
> If required, you can add those fields at [app_mqtt.cpp](https://github.com/Xinyuan-LilyGO/T-ZigBee/blob/main/examples/zigbee2mqtt/app_mqtt.cpp) as follows.
>
> ```
> …
> extern String mqtt_server;
> extern [...]

Thank you very much for providing this tip [swkim01] since most of us have to pass username and password to the mqtt server.  However, when we use this suggested code it causes a panic'ed (Load access fault).  Can you share your complete app_mqtt.cpp with us please?

Also I have found it easier to just edit the ..\T-ZigBee-main\examples\zigbee2mqtt\data\db.json file and adding in your wifi and mqtt details as waiting for the AP and web interface is too slow and buggy...

`{
    "sta": {
        "ssid": "myssid",
        "pwd": "mypassword"
    },
    "zigbee": {
        "channel": 25
    },
    "mqtt": {
        "server": "10.X.X.X",
        "port": 1883
            }`

lbuque, 2022-11-21:
@pawilli 
The ability to use Credentials for MQTT has been added.
3b9ba552b492afed7f105db72d39d0140aa6af11

pawilli, 2022-11-26:
> 

@lbuque 
Thank you for updating the code.  However it does not work...
a) There is no interface to enter the user and password for the mqtt server

name: ssid
args: XXXXXX
name: password
args: xxxxxx
name: server
args: ##########
name: port
args: 1883

Also when you hard code the data in new fields created in 
...\examples\zigbee2mqtt\data\db.json it does not connect.

E (289762) MQTT_CLIENT: MQTT connect failed

My work-around for now is to hardcode the user and password in the mqtt URI line in 
...\examples\zigbee2mqtt\app_mqtt.cpp line 105
replace 
 `sprintf(uri,` "mqtt://%s:%d", mqtt_server.c_str(), `mqtt_port);`

with

`"mqtt://username:password@###.##.#.###:1883"`

[image]

lbuque, 2022-11-26:
@pawilli you need to re-flash littlefs.

pawilli, 2022-11-26:
> @pawilli you need to re-flash littlefs.

@lbuque 

Thank you for your quick reply.

Confirmed working.

Notes for others who need help re-flashing littlefs...

Run the following from the terminal in platformio after you clean and build the new platformio.ini:
```
pio run -e zigbee2mqtt -t upload
pio run -e zigbee2mqtt -t uploadfs

```

Don't forget to hardcode your mqtt username and password in ...\examples\zigbee2mqtt\data\db.json