Xinyuan-LilyGO/T-ZigBee #17: What are the different firmware images distributed for the tlsr8258?

AskAlice, 2022-10-08:
there are 3 firwmwares listed:

[firmware/hciDevice_ZE_8258_SDK_eb644f1_MD5_a27ce68a94bc19284c5c87a4e0aa416b.bin](firmware/hciDevice_ZE_8258_SDK_eb644f1_MD5_a27ce68a94bc19284c5c87a4e0aa416b.bin)
[firmware/hciDevice_ZR_8258_SDK_eb644f1_MD5_3e7c395bacca28475952808a79ff2ec1.bin](firmware/hciDevice_ZR_8258_SDK_eb644f1_MD5_3e7c395bacca28475952808a79ff2ec1.bin)
[firmware/sampleGW_8258_20220302.bin](firmware/sampleGW_8258_20220302.bin)

What does each one do? How are the different HCI device firmware different? Do you have source code for these firmwares? I would like to understand how the esp32 and the tlsr8258 communicate with eachother, and seeing the firmware source would help

wes58, 2022-10-09:
Have a look there - telink-zigbee-sdk [https://github.com/Xinyuan-LilyGO/telink-zigbee-sdk/tree/master/tl_zigbee_sdk/apps](url)

liangyingy, 2022-10-11:
[hciDevice_ZE_8258_SDK_eb644f1_MD5_a27ce68a94bc19284c5c87a4e0aa416b.bin](https://github.com/Xinyuan-LilyGO/T-ZigBee/blob/main/firmware/hciDevice_ZE_8258_SDK_eb644f1_MD5_a27ce68a94bc19284c5c87a4e0aa416b.bin) -> https://github.com/Xinyuan-LilyGO/telink-zigbee-sdk/tree/master/tl_zigbee_sdk/apps/hciDevice

[hciDevice_ZR_8258_SDK_eb644f1_MD5_3e7c395bacca28475952808a79ff2ec1.bin](https://github.com/Xinyuan-LilyGO/T-ZigBee/issues/firmware/hciDevice_ZR_8258_SDK_eb644f1_MD5_3e7c395bacca28475952808a79ff2ec1.bin) -> https://github.com/Xinyuan-LilyGO/telink-zigbee-sdk/tree/master/tl_zigbee_sdk/apps/hciDevice

[sampleGW_8258_20220302.bin](https://github.com/Xinyuan-LilyGO/T-ZigBee/blob/main/firmware/sampleGW_8258_20220302.bin) -> https://github.com/Xinyuan-LilyGO/telink-zigbee-sdk/tree/master/tl_zigbee_sdk/apps/sampleGw

closed 2022-10-11 by liangyingy