Xinyuan-LilyGO/T-ZigBee #21: Does anyone have a good guide ?

ausratty, 2023-01-22:
Hey guys, i've spent way, way too much time trying to get the zigbee board to work. I've followed the youtube video Academia  has put up. However i get issues with esp_littlefs mount failed once i switch back to the esp to load the mqtt image. 
I'm at a loss, i've got 30 of these that are about to go in the bin, looks like we'll have to go back to esp32s. But we really need the zigbee alternative.

pawilli, 2023-01-24:
> Hey guys, i've spent way, way too much time trying to get the zigbee board to work. I've followed the youtube video Academia has put up. However i get issues with esp_littlefs mount failed once i [...]

Hi @ausratty,

What are you trying to do?

I have got all of the examples to work and I have been running the zibgee light example for a while now on three boards as battery backed up routers across my network.

I have got the gateway example to work with Homeassistant too but it has limited utility as @lbuque has not shared how to add more devices to the supported list.

Let me know what you want to do and I will try to assist.

pawilli, 2023-01-24:
> Hey guys, i've spent way, way too much time trying to get the zigbee board to work. I've followed the youtube video Academia has put up. However i get issues with esp_littlefs mount failed once i [...]

See this link for some examples
https://github.com/Xinyuan-LilyGO/T-ZigBee/issues/19#issue-1514503479

ausratty, 2023-01-25:
Hey @pawilli I've just been trying to load the sample firmware on it and adopt into HA, but not getting anywhere. I was simply hoping to use them to fire relays for irrigation. 
I've followed all the guides and always seem to get to a point where it won't boot and has a FS issue.

pawilli, 2023-01-25:
> Hey @pawilli I've just been trying to load the sample firmware on it and adopt into HA, but not getting anywhere. I was simply hoping to use them to fire relays for irrigation. I've followed all the guides and always seem to get to a point where it won't boot and has a FS issue.

Are you using platformio and Visual Sudio Code?

Ok assuming you have uploaded the correct firmware to the TLSR8258 using the T-U2T and appropriate pin configuration - 

Do a "clean" then "build" the firmware and "upload" it to the board (don't forget to reset the pins for the ESP32)

Then after you upload the firmware to the board run the following commands in the terminal

`pio run -e zigbee2mqtt -t upload`
`pio run -e zigbee2mqtt -t uploadfs`

To re-flash littlefs... (assuming you are trying to load the gateway example.)

See this thread for more information https://github.com/Xinyuan-LilyGO/T-ZigBee/issues/16#issuecomment-1328002824

adoucette, 2023-05-26:
I am also trying to get started using this board. Would like to build a STEM project together with my son and then adopt it into HA with zigbee.
However, I don't know where to start.
Is there a driver that I need to connect my PC to the board with the T-U2T adapter? Which driver?
Then how do I go about uploading the firmware(s)?
Once that's done I think there are some guides around the internet on how to program the esp32 relative to the sensors project we'd like to do.

adoucette, 2023-05-28:
It would be useful to start a collaborative guide on how to get up and running with this Lilygo T-Zigbee device.  
Personally, I am not familiar with these ESP32 devices and it is taking me a good amount of time to find the relevant steps from a lot of different places - just to get started. I would like to save others that trouble, as well as to figure out for myself what I might be missing or doing wrong.  
I would like to get Zigbee up and running -- I think that's pretty much why people buy this particular board. In my case I'd like to connect it to Home Assistant and I don't really care whether it needs MQTT (which is already running on my HA server) or can just connect through any other Zigbee extension on HA.  
Regardless, in case it helps others, here's what I've had to do so far. I'm using Windows.

1.  Install Visual Studio Code
2.  Install Platformio extension in VS Code (press Ctrl + Shift +X to bring up the Extensions window, then search for PlatformIO)
3.  Install C/C++ Extension Pack in VS Code (as above)
4.  Install Git Extension Pack in VS Code (as above)
5.  Install the silabs drivers on your computer, such as the "[CP210x\_Universal\_Windows\_Driver](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads)" (Windows users: there will be a "silabser.inf" file.  Right click on this and select "install")
6.  Now you will (finally) be able to connect the device via USB-C cable and [LilyGo's "T-U2T" adapter](https://www.lilygo.cc/products/t-u2t). Ensure the device shows up in device manager (if not you may be a using bad USB cable - trust me and try different cables). When connected, note the COM port number.
7.  In VS Code, click on the PlatformIO icon (looks like a bug/alien head) and choose the option to make a project. Your project name should have no spaces.  
    The "Board" should be "TTGO T-OI PLUS RISC-V ESP32-C3". Framework is "Arduino"
8.  Then in PlatformIO's platformio.ini file, make sure you have the following:
    1.  \[env:ttgo-t-oi-plus\]  
        platform = espressif32  
        board = ttgo-t-oi-plus  
        framework = arduino  
        board\_build.mcu = esp32c3  
        board\_build.f\_cpu = 160000000L  
        upload\_speed = 921600  
        monitor\_speed = 115200  
        lib\_ignore = examples  
        lib\_deps =  
            https://github.com/Xinyuan-LilyGO/T-ZigBee.git  
            bblanchon/ArduinoJson@^6.19.4  
            mathertel/OneButton@^2.0.3  
        board\_build.filesystem = littlefs
9.  Then you'll be ready to follow this example: [https://github.com/Xinyuan-LilyGO/T-ZigBee/tree/main/examples/factory_test](https://github.com/Xinyuan-LilyGO/T-ZigBee/tree/main/examples/factory_test)
    1.  Note you will have to make all of the .h files in platformio under includes and all of the .c files under source.
    2.  Once all those are added, edit the source code to configure the wifi for your local network (don't worry about configuring the sparkfun account). Save all open tabs.
    3.  Then click the build icon in PlatformIO (looks like a checkmark on the lower status bar of the program window). Correct errors as needed. When you have a successful build, move on.
    4.  Then use your mouse to click on the transfer icon in platformio (program window lower status bar, button looks like a right-arrow) and immediately use your finger to long press the KEY button on your device. Once the terminal output in platformio indicates it is transferred to your device successfully, use your finger to click the reset button on your device. The device should boot and should connect to your specified wifi network.
    5.  Take a look at the terminal output in Visual Studio Code. It should indicate whether or not you've completed this successfully.
10.  So, that simple example done, use the above technique to then follow this example to update the firmware on the zigbee chip (the TLSR8258):  
    [https://zbhci.readthedocs.io/en/latest/user-guide/burning.html](https://zbhci.readthedocs.io/en/latest/user-guide/burning.html)
    1.  Get the most up-to-date firmware from this git repository.
    2.  Be sure to flip the dip switches per the guide when called on to do so.
    3.  Update the TLSR8258 firmware with the TlsrComSwireWriter linked to from that page. To do so, just download the whole zip of the updater source from github and extract to your computer. Then open the html file in that folder (this seems to work better in chrome). Use this web GUI to update the firmware.

From there I moved on to the zigbee2mqtt example, but haven't yet been able to get it to connect to Home Assistant using Zigbee. Still working on that.

jrossouw, 2023-09-11:
@adoucette it is a briliant plan to start a community page where we can all share information like this. Any ideas where we can put it? Wikipedia perhaps?