Xinyuan-LilyGO/T-ZigBee #22: Add ESP32C6.Light device

lbuque, 2023-02-10:
Hardware: [ESP32-C6-DevKitC-1](https://item.taobao.com/item.htm?ft=t&id=695606483841)
Software: [esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_on_off_ligh](https://github.com/espressif/esp-zigbee-sdk/tree/1a52786a4dd972c153ffd77e4da8a3d33fa4eaac/examples/esp_zigbee_HA_sample/HA_on_off_light)
esp-zigbee-sdk: [1a52786a4dd972c153ffd77e4da8a3d33fa4eaac](https://github.com/espressif/esp-zigbee-sdk/tree/1a52786a4dd972c153ffd77e4da8a3d33fa4eaac)
esp-idf: [76433e4cb22a1d88ed505b2abba1e56f1b7241ff](https://github.com/espressif/esp-idf/tree/76433e4cb22a1d88ed505b2abba1e56f1b7241ff)

Here is the precompiled bin file: [bin.tar.gz](https://github.com/Xinyuan-LilyGO/T-ZigBee/files/10705038/bin.tar.gz)

Burn it with the following command:

```
esptool esp32c6 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 on_off_light_bulb.bin 0x8000 partition_table/partition-table.bin
```

Add the following patches to esp-zigbee-sdk:

```diff
diff --git a/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c b/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c
index 4159c93..3459851 100644
--- a/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c
+++ b/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c
@@ -85,15 +85,56 @@ void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
     }
 }
 
+char modelid[] = {13, 'E', 'S', 'P', '3', '2', 'C', '6', '.', 'L', 'i', 'g', 'h', 't'};
+char manufname[] = {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'};
+
 static void esp_zb_task(void *pvParameters)
 {
     /* initialize Zigbee stack with Zigbee end-device config */
     esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
     esp_zb_init(&zb_nwk_cfg);
+    esp_zb_set_network_channel(25);
+
+
     /* set the on-off light device config */
-    esp_zb_on_off_light_cfg_t light_cfg = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
-    esp_zb_ep_list_t *esp_zb_on_off_light_ep = esp_zb_on_off_light_ep_create(HA_ESP_LIGHT_ENDPOINT, &light_cfg);
-    esp_zb_device_register(esp_zb_on_off_light_ep);
+    uint8_t basic_version, basic_power_source;
+    uint8_t test_attr = 0;
+
+    basic_version = ESP_ZB_ZCL_BASIC_ZCL_VERSION_DEFAULT_VALUE;
+    basic_power_source = ESP_ZB_ZCL_BASIC_POWER_SOURCE_DEFAULT_VALUE;
+    /* basic cluster create with fully customized */
+    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
+    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &basic_version);
+    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &basic_power_source);
+    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);
+    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);
+    /* identify cluster create with fully customized */
+    esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY);
+    esp_zb_identify_cluster_add_attr(esp_zb_identify_cluster, ESP_ZB_ZCL_ATTR_IDENTIFY_IDENTIFY_TIME_ID, &test_attr);
+    /* group cluster create with fully customized */
+    esp_zb_attribute_list_t *esp_zb_groups_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_GROUPS);
+    esp_zb_groups_cluster_add_attr(esp_zb_groups_cluster, ESP_ZB_ZCL_ATTR_GROUPS_NAME_SUPPORT_ID, &test_attr);
+    /* scenes cluster create with standard cluster + customized */
+    esp_zb_attribute_list_t *esp_zb_scenes_cluster = esp_zb_scenes_cluster_create(NULL);
+    esp_zb_cluster_update_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_NAME_SUPPORT_ID, &test_attr);
+    /* on-off cluster create with standard cluster config*/
+    esp_zb_on_off_cluster_cfg_t on_off_cfg;
+    on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
+    esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);
+    /* create cluster lists for this endpoint */
+    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
+    esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
+    /* update basic cluster in the existed cluster list */
+    //esp_zb_cluster_list_update_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
+    esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
+    esp_zb_cluster_list_add_groups_cluster(esp_zb_cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
+    esp_zb_cluster_list_add_scenes_cluster(esp_zb_cluster_list, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
+    esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
+
+    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
+    /* add created endpoint (cluster_list) to endpoint list */
+    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID);
+    esp_zb_device_register(esp_zb_ep_list);
     esp_zb_device_add_set_attr_value_cb(attr_cb);
     ESP_ERROR_CHECK(esp_zb_start(false));
     esp_zb_main_loop_iteration();
diff --git a/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.h b/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.h
index d909edf..2d53a2f 100644
--- a/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.h
+++ b/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.h
@@ -19,7 +19,7 @@
 #define INSTALLCODE_POLICY_ENABLE       false    /* enable the install code policy for security */
 #define ED_AGING_TIMEOUT                ESP_ZB_ED_AGING_TIMEOUT_64MIN
 #define ED_KEEP_ALIVE                   3000    /* 3000 millisecond */
-#define HA_ESP_LIGHT_ENDPOINT           10    /* esp light bulb device endpoint, used to process light controlling commands */
+#define HA_ESP_LIGHT_ENDPOINT           1    /* esp light bulb device endpoint, used to process light controlling commands */
 
 #define ESP_ZB_ZED_CONFIG()                                         \
     {                                                               \

```

Thanks to @swkim01  for the help!!!

joshuajonah, 2023-02-12:
Wait, so this is a commit to the esp32-h1 SDK? This is tied to the T-Zigbee codebase?

lbuque, 2023-02-12:
@joshuajonah This is to add ESP32-C6 as a zigbee sub-device to the T-ZigBee gateway.

pawilli, 2023-02-14:
> 

Thank you @lbuque but these installation instructions are not clear for a novice beginner like me.

You suggest - 

1. that we burn the precompiled bin file using esptool to esp32c6 which means we need to set the jumpers to the esp32 configuration and burn right?
2. Patch the esp-zigbee-sdk - can you provide some guidance on how to do this and in which environment?  And do we have to generate a new firmware for the TLSR8258?
3. Or do we just update from the current (new) commit and rerun the zigbee2mqtt example?

Thank you

pawilli, 2023-02-14:
> Hardware: [ESP32-C6-DevKitC-1](https://item.taobao.com/item.htm?ft=t&id=695606483841) Software: [esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_on_off_ligh](https://github.com/espressif/esp-zigbee-sdk/tree/1a52786a4dd972c153ffd77e4da8a3d33fa4eaac/examples/esp_zigbee_HA_sample/HA_on_off_light) esp-zigbee-sdk: [1a52786a4dd972c153ffd77e4da8a3d33fa4eaac](https://github.com/espressif/esp-zigbee-sdk/tree/1a52786a4dd972c153ffd77e4da8a3d33fa4eaac) esp-idf: [76433e4cb22a1d88ed505b2abba1e56f1b7241ff](https://github.com/espressif/esp-idf/tree/76433e4cb22a1d88ed505b2abba1e56f1b7241ff)
> 
> Here is the precompiled bin file: [bin.tar.gz](https://github.com/Xinyuan-LilyGO/T-ZigBee/files/10705038/bin.tar.gz)
> 
> Burn it with the following command:
> 
> ```
> esptool esp32c6 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 on_off_light_bulb.bin 0x8000 partition_table/partition-table.bin
> ```
> ">>>SNIPPED<<<"
> Thanks to @swkim01 for the help!!!

Also trying to run your command to burn the precompiled bin file produces an error - 

`esptool esp32c6 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 on_off_light_bulb.bin 0x8000 partition_table/partition-table.bin`

Modified the command for my env to 

`esptool --chip esp32c6 -p COM6 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 on_off_light_bulb.bin 0x8000 partition_table/partition-table.bin`

But still got an error...
`A fatal error occurred: This chip is ESP32-C3 not ESP32-C6. Wrong --chip argument?`

Modified the command to esp32c3 and got another error

```
esptool --chip esp32c3 -p COM6 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 on_off_light_bulb.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.3
Serial port COM6
Connecting....
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: XX:XX:XX:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Unexpected chip id in image. Expected 5 but value was 13. Is this image for a different chip model?

A fatal error occurred: bootloader/bootloader.bin is not an ESP32-C3 image. Use --force to flash anyway. 

```

lbuque, 2023-02-14:
This bin file is for esp32c6, did my above words make you misunderstand?

pawilli, 2023-02-14:
> @pawilli This bin file is for esp32c6, did my above words make you misunderstand?

Oh I see - so maybe this was probably posted to the wrong forum.  I thought this was for the t-zigbee board with a ESP32-C3 chip