Xinyuan-LilyGO/T-ZigBee #25: Understanding ZclAttrRead flow

Osendo, 2023-03-13:
Hi. I'm new to the zigbee stack, I'm having some trouble with a request to read an attribute from a RTCGQ11LM sensor. 
Maybe I don't understand the request path correctly, in that case please correct me.
Example request:

```cpp
uint16_t au16AttrList[1] = {0x0000};
zbhci_ZclAttrRead(0x03, (ts_DstAddr) {.u64DstAddr = deviceList[i].u64IeeeAddr}, 0, 1, 0, 0x0406, 1, au16AttrList);
```

I expect a message to come from the sensor after this with u16MsgType == ZBHCI_CMD_ZCL_ATTR_READ_RSP

lbuque, 2023-03-13:
You can try to wake it up by pressing the button of RTCGQ11LM within 0-7 seconds after sending zbhci_ZclAttrRead.

ps: The RTCGQ11LM sensor is a dormant device, it is only active for one minute after joining the network.

Osendo, 2023-03-13:
<details>
  <summary>hci log</summary>
  
[ 81821][I][hci_uart.c:126] uart_event_task(): uart[1] event:
[ 81850][I][hci_uart.c:134] uart_event_task(): [UART DATA]: 17
[ 81855][I][hci_display.c:371] displayZclAttrReadRsp(): Type: 0x8100
[ 81861][I][hci_display.c:372] displayZclAttrReadRsp():   (ZCL Attr Read Rsp)
[ 81868][I][hci_display.c:373] displayZclAttrReadRsp():   SQN:        0x7
[ 81875][I][hci_display.c:374] displayZclAttrReadRsp():   Src Addr:   0x2a71
[ 81881][I][hci_display.c:375] displayZclAttrReadRsp():   Src Ep:     0x1
[ 81888][I][hci_display.c:376] displayZclAttrReadRsp():   Cluster ID: 0x406
[ 81894][I][hci_display.c:377] displayZclAttrReadRsp():   Attr Num:   0x01
[ 81901][I][hci_display.c:380] displayZclAttrReadRsp():   Attr List:
[ 81907][I][hci_display.c:383] displayZclAttrReadRsp():     Attr 0:
[ 81913][I][hci_display.c:384] displayZclAttrReadRsp():       Attr ID:   0000
[ 81920][I][hci_display.c:385] displayZclAttrReadRsp():       Data Type: 0x16
[ 81927][I][hci_display.c:386] displayZclAttrReadRsp():       Data Len:  0000
[ 82532][I][hci_uart.c:126] uart_event_task(): uart[1] event:
[ 82533][I][hci_uart.c:134] uart_event_task(): [UART DATA]: 10
[ 82533][I][hci_display.c:762] displayDataConfirm(): Type: 0x8200
[ 82538][I][hci_display.c:763] displayDataConfirm():   (Data Confirm)
[ 82544][I][hci_display.c:764] displayDataConfirm():   Status:  00
[ 82539][I][hci_uart.c:126] uart_event_task(): uart[1] event:
[ 82556][I][hci_uart.c:134] uart_event_task(): [UART DATA]: 17
[ 82550][I][hci_display.c:765] displayDataConfirm():   Src Ep:  0x1
[ 82567][I][hci_display.c:766] displayDataConfirm():   Aps Cnt: 0xf3
[ 82573][I][hci_display.c:371] displayZclAttrReadRsp(): Type: 0x8100
[ 82579][I][hci_display.c:372] displayZclAttrReadRsp():   (ZCL Attr Read Rsp)
[ 82586][I][hci_display.c:373] displayZclAttrReadRsp():   SQN:        0x8
[ 82593][I][hci_display.c:374] displayZclAttrReadRsp():   Src Addr:   0x2a71
[ 82600][I][hci_display.c:375] displayZclAttrReadRsp():   Src Ep:     0x1
[ 82606][I][hci_display.c:376] displayZclAttrReadRsp():   Cluster ID: 0x406
[ 82613][I][hci_display.c:377] displayZclAttrReadRsp():   Attr Num:   0x01
[ 82619][I][hci_display.c:380] displayZclAttrReadRsp():   Attr List:
[ 82625][I][hci_display.c:383] displayZclAttrReadRsp():     Attr 0:
[ 82631][I][hci_display.c:384] displayZclAttrReadRsp():       Attr ID:   0000
[ 82638][I][hci_display.c:385] displayZclAttrReadRsp():       Data Type: 0x16
[ 82645][I][hci_display.c:386] displayZclAttrReadRsp():       Data Len:  0000
[ 86265][I][hci_uart.c:126] uart_event_task(): uart[1] event:
[ 86265][I][hci_uart.c:134] uart_event_task(): [UART DATA]: 11
[ 86265][I][hci_display.c:48] displayAcknowledg(): Type: 0x8000
[ 86271][I][hci_display.c:49] displayAcknowledg():   (Status)
[ 86276][I][hci_display.c:50] displayAcknowledg():   Status:  00
[ 86282][I][hci_display.c:51] displayAcknowledg():   Message: 0x100
[ 91265][I][hci_uart.c:126] uart_event_task(): uart[1] event:
[ 91265][I][hci_uart.c:134] uart_event_task(): [UART DATA]: 11
[ 91265][I][hci_display.c:48] displayAcknowledg(): Type: 0x8000
[ 91270][I][hci_display.c:49] displayAcknowledg():   (Status)
[ 91276][I][hci_display.c:50] displayAcknowledg():   Status:  00
[ 91282][I][hci_display.c:51] displayAcknowledg():   Message: 0x100
[ 96154][I][hci_uart.c:126] uart_event_task(): uart[1] event:
[ 96154][I][hci_uart.c:134] uart_event_task(): [UART DATA]: 10
[ 96154][I][hci_display.c:762] displayDataConfirm(): Type: 0x8200
[ 96160][I][hci_display.c:763] displayDataConfirm():   (Data Confirm)
[ 96166][I][hci_display.c:764] displayDataConfirm():   Status:  0xf0
[ 96172][I][hci_display.c:765] displayDataConfirm():   Src Ep:  0x1
[ 96178][I][hci_display.c:766] displayDataConfirm():   Aps Cnt: 0xf4
</details>
As I understand it, I got a response from the device that it accepted the read request, but nothing happened :)
The device is alive and sends data when a key is pressed or movement

lbuque, 2023-03-14:
> Data Type: 0x16

This data type is invalid.

The supported data types are defined here.

https://github.com/Xinyuan-LilyGO/T-ZigBee/blob/main/src/zbhci_commom.h#L18-L77

asat232, 2024-11-01:
0x16 is Logical Reserved Data Type.
And i observed behaviour like this plenty of times, not all attributes about those notifying sensor are supposed to be read (or write), and when attribute on sensor side not implemented for read, then sensor responding with Reserved Data Type or even with wrong attribute ID.