Xinyuan-LilyGO/T-ZigBee #26: Create light control with 2 endpoint

settler-mar, 2023-03-18:
I create firmware and send status of 2 channel to zigbee2mqtt
- zbhci_ZclSendReportCmd(0x02, sDstAddr, 1, 1, 0, 1, 0x0006, 0x0000, ZCL_DATA_TYPE_BOOLEAN, 1, &ledState);
- zbhci_ZclSendReportCmd(0x02, sDstAddr, 2, 1, 0, 1, 0x0006, 0x0000, ZCL_DATA_TYPE_BOOLEAN, 1, &ledState);
It`s work 

I create config file with 2 endpoint In zigbee2mqtt
UI interface:
[image]

Endpoint l1 work normal (send to/from device)
Endpoint l2 send only from device to UI. Whern I click to swich button in UI I get error after 1s
[image]
Publish 'set' 'state' to '0xa4c138c681c6b987' failed: 'Error: Command 0xa4c138c681c6b987/2 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51948 - 2 - 6 - 6 - 11 after 10000ms)'

lbuque, 2023-03-19:
@settler-mar Because the firmware of tlsr8258 only exists endpoint1, endpoint2 is not defined.

Only endpoint1 is defined here:
https://github.com/Xinyuan-LilyGO/telink-zigbee-sdk/blob/master/tl_zigbee_sdk/apps/hciDevice/sampleGateway.c#L172

settler-mar, 2023-03-19:
How can i fix it?