Xinyuan-LilyGO/T-ZigBee #28: Can this board act as slave?

andreapx, 2023-05-14:
Can I connect it to my existing ZigBee network and connect to it some relays?
Where can I see some examples?
Thanks!

andreapx, 2024-01-25:
Sorry, I'll start over: I've had a look at the documentation, but nowhere I see how to use it. How do I connect the sensors? How do I connect the relays? How do I edit the configuration to use it with temperature sensors (or other) or with relays?
Thanks!

papadeltasierra, 2024-01-25:
As to the actual electronics, I can only say that you have two options:
- Have the tlsr8258 chip react directly and use its pins (marked PA/B/C/D) to drive stuff
- Have the tlsr8258 and ESP32c3 talk using the Telink zbhci interface and then have the ESP32c3 GPIO pins drive stuff.
You will find far more on the internet about interfacing the ESP32c3 so I'd go that route.  You then treat each thing as being driven by/driving-or-setting a ZigBee attribute.
If you don't understand ZigBee yet, read up on how it creates networks and then how attributes are read and written.  You "write" to drive things and "read" to monitor things and your sensor/controller device sets or uses the attribute value to report information of drive things like relays.
ZigBee has many attributes, organises into "Clusters" - expect a longish journey getting up to speed ;-).