Xinyuan-LilyGO/T-ZigBee #29: Please help connecting to Home Assistant

adoucette, 2023-05-28:
How do I go about connecting the device to Home Assistant by Zigbee or MQTT?  
I believe I have built the zigbee2mqtt example and uploaded it to the LilyGo T-Zigbee device.  
Prior to building, I made the following edits in the code:

1\. data/db.json: set the wifi router ssid and pwd, set the zigbee channel to what my MQTT server is using (15), set the MQTT server ip, port, username, and password  
2\. zigbee2mqtt.ino: set the wifi router ssid and password, the mqtt server ip, port user, and password; set the "zbhci\_BdbChannelSet" to my MQTT server channel (15)

Then in VS Code I can build and upload it to the device. Then in the VS Code terminal window I see that the device connects to my wifi, but is not connecting to MQTT. I have held down the "Key" button on the device to set it into pairing mode (blue light on chip opposite corner from the USB input is slow flashing).

Here is some output from the terminal window:

```cpp
--- Terminal on COM6 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x91c
load:0x403ce710,len:0x25b0
entry 0x403cc710
WiFi: Set mode to WIFI_AP_STA
Waiting for WiFi
WiFi: softAP has been established
WiFi: please connect to the LILYGO-8821
HTTP server started
E (1289) MQTT_CLIENT: Client was not initialized
Nothing saved yet!
[WiFi-event] event: 4
Connected to access point
[WiFi-event] event: 7
Obtained IP address: 192.168.30.153
E (1372) MQTT_CLIENT: MQTT connect failed
E (16395) MQTT_CLIENT: MQTT connect failed
E (31407) MQTT_CLIENT: MQTT connect failed
E (46511) MQTT_CLIENT: MQTT connect failed
E (61521) MQTT_CLIENT: MQTT connect failed
```

In HA I don't see it as a zigbee device to add.

How do I get this connected to HA?

EKazakof, 2023-09-01:
Without understanding anything, I bought 3 modules. Now this garbage is lying in a box and waiting for the manufacturer to write at least something that works.

adoucette, 2023-09-02:
Same. Looking forward to ESPHome support for the latest version chips since some of those have Zigbee and Thread connectivity.

buegelbeatz, 2024-06-29:
Maybe this could help: https://github.com/Xinyuan-LilyGO/T-ZigBee/pull/24 - No clue why this simple easy pull request is never merged