Xinyuan-LilyGO/T-ZigBee #33: Setting the manufacturerName not working

jrossouw, 2023-09-09:
The example code does not appear to set the manufacturer code correctly when pairing the device with my Homey Pro controller. My controller always show the manufacturer as "TELINK" while in the example sensor code it is set it to:

`
const uint8_t au8ManufacturerName[] = {13,'L','I','L','Y','G','O', '.', 'S', 'e', 'n', 's', 'o', 'r'};
`

Has anyone else experienced this issue?

papadeltasierra, 2024-01-25:
You need to lok at the example code that sets
```
zcl_basicAttr_t g_zcl_basicAttrs = ...
```
This pulls in a `#defined` byte array that should contain your manufacturer name and the model.  This definitely works as I am using it.

asat232, 2024-10-08:
Absolutely correct question.
First of all lets look how its intended to work for example zigbee2mqtt (as coordinator) and sampleLight (as router).
Both of them having two parts of code: first for ESP32C3 in this repo and pre-built for TLSR8258 <a href="https://github.com/Xinyuan-LilyGO/telink-zigbee-sdk">telink-zigbee-sdk</a> (slightly modified version of original <a href="https://github.com/telink-semi/telink_zigbee_sdk">telink_zigbee_sdk</a>).

zigbee2mqtt (ccoridnator) getting Announcement Indication <a href="https://github.com/Xinyuan-LilyGO/T-ZigBee/blob/main/examples/zigbee2mqtt/zigbee2mqtt.ino#L408-L409">code here</a>
<pre><code>uint16_t au16AttrList[1] = {0x0005};
zbhci_ZclAttrRead(0x02, (ts_DstAddr) {.u16DstAddr = payload->u16NwkAddr}, 1, 1, 0, 0x0000, 1, au16AttrList);</code></pre>
And getting respose from sampleLight (router):
<code><pre>ZCL_ATTR_READ_RSP
Type: 0x8100
  (ZCL Attr Read Rsp)
  Src Ep:     0x1
  Cluster ID: 0000
  Attr Num:   0x01
  Attr List:
    Attr 0:
      Attr ID:   0x05
      Data Type: 0x42
      Data Len:  0x08
      Attr Data: TLSR82xx</code></pre>
because in pre-built binary we have this <a href="https://github.com/Xinyuan-LilyGO/telink-zigbee-sdk/blob/master/tl_zigbee_sdk/apps/sampleLight/sampleLightEpCfg.c#L208-L209">code</a>
But on successful commissioning <a href="https://github.com/Xinyuan-LilyGO/telink-zigbee-sdk/blob/master/tl_zigbee_sdk/apps/hciDevice/zb_appCb.c#L189">code here</a>, sampleLight getting ZBHCI_CMD_NETWORK_STATE_REPORT <a href="https://github.com/Xinyuan-LilyGO/T-ZigBee/blob/main/examples/light_demo/light_demo.ino#L144-L145">code here</a>
<pre><code>sDstAddr.u16DstAddr = 0x0000;
zbhci_ZclSendReportCmd(0x02, sDstAddr, 1, 1, 0, 1, 0x0000, 0x0005, ZCL_DATA_TYPE_CHAR_STR, sizeof(au8ManufacturerName), (uint8_t *)&au8ManufacturerName);</code></pre>
Coordinator getting:
<pre><code>ZCL_REPORT_MSG_RCV
Type: 0x8104
  (ZCL Attr Report)
  Src Ep:     0x1
  Cluster ID: 0000
  Attr Num:   0x01
  Attr List:
    Attr 0:
      Attr ID:   0x05
      Data Type: 0x42
      Data Len:  0x0c
      Attr Data: LILYGO.Light</code></pre>
So, if coordinator reading Basic Cluster Attributes its always getting what is hardcoded in TLSR8258 firmware.
The code in sampleLight NOT setting Attribute, its just sending Attribute Report with fake value.