Xinyuan-LilyGO/T-ZigBee #44: End Devices selection help

asat232, 2024-10-04:
I need End Devices like small motion sensor and door sensor, and to make them working with coordinator based on simpleGW firmware. Which brand of sensors better to select?
Googling, i found one reference that Sonoff sensors using inside TLSR828 chip.
So, if i use  Sonoff SNZB-03 (motion sensor) and Sonoff SNZB-04 (door sensor), will they speak with coordinator based on simpleGW firmware?
I understand i will need to modify zigbee2mqtt, the only thing i'm worrying  the need to modify simpleGW firmware.

Which brand and models of sensors better to select for adaptation?

asat232, 2024-10-17:
i bought 3 sensors:
Sonoff SNZB-03 (motion sensor)
Sonoff SNZB-03P (motion sensor)
Sonoff SNZB-04 (door sensor)
But ccordinator receiving nothing from them.
Any idea?

asat232, 2024-11-01:
The answer is, different protocol stack. Now i'm using with T-ZigBee coordinator tuya compatible sensors, but i'm not using simpleGW firmware, nor mqtt. T-ZigBee is great board, but i can not say this about documentation. To understand how all this stuff working i used NPX and TELINK documentation.

closed 2024-11-01 by asat232

mr-bobur, 2024-12-12:
so how is your work going, we are currently trying to deploy the telink gataway code as recognizing new tyepes of devices

mr-bobur, 2024-12-12:
Do you try it out before i mean, did you try with telink ide and t-zigbee project

asat232, 2024-12-12:
yes, i used <a href="https://github.com/Xinyuan-LilyGO/telink-zigbee-sdk">github.com/Xinyuan-LilyGO/telink-zigbee-sdk</a> and built sampleGW.bin , using TELInk <b>OLD</b> IDE as this SDK is slightly modified 3.6.6.0 version of TELINK original SDK <a href="https://github.com/telink-semi/telink_zigbee_sdk">github.com/telink-semi/telink_zigbee_sdk</a> (which is dated middle 2021).
After this i added commands to coordinator (gateway) to suite my needs.
I thought my project going to finish. But two days ago i found CRITICAL bug.
T-ZigBee programmed as router not working as router, routers i expected to use as expanders for sensors.
Looking the way its done, i understood, no sense to fix this  based on v3.6.6.0 SDK.
Now i'm trying to adopt latest v3.7.7.1 (dated 2024) original TELINK SDK using their IDE 2024.