Xinyuan-LilyGO/T-ZigBee #46: tlsr8258 to recognizing new device ?

mr-bobur, 2024-12-10:
i have few tuya sensors like water leakage motion and human presence sensors, so now i adding these devices to my zigbee network and listen but the tlsr sending exact one msg to esp32, the msg is recive confirmation msg with  0x8200 code. 
How can i add these devices to tlsr recognition or can i manage it with esp32 ???

asat232, 2024-12-11:
so i understand tuya sensors ZigBee protocol stack is compatible with simpleGW firmware of TLSR8258, because coordinator (T-ZigBee board) working for me with tuya sensor models: TS0203 (door sensor), TS0202 (pir+light sensor) of different manufacturers.
I tried to use Sonoff sensors <a href="https://github.com/Xinyuan-LilyGO/T-ZigBee/issues/44">End Devices selection help</a>, but my coordinator did not get anything from them.
So i remember, msg with 0x8200 code, is just confirmation that TLSR8258 received command from esp32 and send it.
 But my approach was: 
first i made working two T-ZigBee boards one as coordinator, another as router (to be sure my coordinator working correctly).
Then, begin to work with sensors.

asat232, 2024-12-22:
you was interested how to build TLSR firmware using TELINK  ide.
Just for information.
v3.7.1.1 original TELINK ZigBee SDK.
IDE:   Telink IoT Studio 2024.8
 For me, build of sampleGW project was working out of box,
Certainly you need configure project:
<code>#define	ZBHCI_UART 1              //file app_cfg.h
#define DONGLE_8258_32	   1   //file board_8258_dongle.h
#define	DONGLE_8258_48  0   //file board_8258_dongle.h</code>

You may ask why no original Telink SDK modifications, because instead of custom HCI functions, i used original Telink SDK functions:
ZBHCI_CMD_NETWORK_STATE_REQ //custom SDK
ZBHCI_CMD_GET_LOCAL_NWK_INFO_REQ // original TelinkSDK, same even more extended info

ZBHCI_CMD_NETWORK_STATE_RSP //custom SDK
ZBHCI_CMD_GET_LOCAL_NWK_INFO_RSP// original TelinkSDK, same even more extended info

- i'm not using sending back and forth device-model attribute by custom functions, i'm getting device-model attribute as it is, from sensors, so no need of modifications.
- custom SDK  is based on original v3.6.6.0 all ZCL functions not included address info, but in new v3.7.1.1 this info is present, so no need for modifications.

Certainly HCI interface on esp32 side you need to modify accordingly new SDK (add params, change order).

Build router is another story. But using v.3.7.1.1 i tested even two routers as distance range extender working:
coordinator<->router<->router<->sensors

asat232, 2024-12-22:
By the way, this days i encountered once after reprogramming TLSR firmware (coordinator) it stopped receiving anything from sensors. Simple solution, just attach long click from button and put in it TLSR reset through HCI (it will reset all underlying layers in TLSR chip). 
<code>zbhci_BdbFactoryReset();</code>
May need only first run. Later you can comment this code.

mr-bobur, 2025-01-03:
> By the way, this days i encountered once after reprogramming TLSR firmware (coordinator) it stopped receiving anything from sensors. Simple solution, just attach long click from button and put in it TLSR reset through HCI (it will reset all underlying layers in TLSR chip). `zbhci_BdbFactoryReset();` May need only first run. Later you can comment this code.

thank you to answer the issue. So i will try to figure out building tlsr sdk for my own solution. could you also share mail to immadaite response,